analogdevicesinc/scopy #412: Signal Generator: Waveform type always loads sinewave on channel 1

drbautista, 2018-04-24:
OS: Windows 10 64bit
 Scopy Version: f2c402a
 Firmware Version: v0.19

Steps to reproduce:

1. Set Channel 1's waveform type to square wave. and channel 2 to triangle wave.
2. Set different amplitude, offset, frequency and phase.
3. Save profile.
4. Reset Scopy on preference menu.
5. Load the previously saved profile.

Channel 1's waveform type is always returning to sinewave.
The bug on channel 2's waveform type is not always present.

claudiupop, 2018-04-26:
Should be fixed with commit 1bfadf59419d5da2f516b40a05f8b5c56f36c1bb

amiclaus, 2018-05-02:
The bug is partially fixed, excepting the number of periods displayed.
[image]

claudiupop, 2018-05-04:
This should be fixed with f0b2eaa48607416f517a4ceb5bdb8d57cd5284c0

amiclaus, 2018-05-07:
On Scopy build: 5af87b4 (v0.12) the bug is fixed.

closed 2018-05-07 by amiclaus