analogdevicesinc/scopy #413: Signal Generator: Buffer's scaling setting is not saved

drbautista, 2018-04-24:
OS: Windows 10 64bit
 Scopy Version: f2c402a
 Firmware Version: v0.19

Steps to reproduce:
1. Change the scaling settings on the buffer tab.
2. Save a profile.
3. Reset Scopy on the preferences menu and reload the previously saved profile.

The set scaling values are returned to default.

*When you load a .csv file before saving the profile, the file's Path, format and Size is also not saved(Though I'm not sure if this is a feature that's added originally.)

claudiupop, 2018-05-04:
Should be fixed with f0b2eaa48607416f517a4ceb5bdb8d57cd5284c0 and now it will also load the previous loaded file.

amiclaus, 2018-05-07:
Scopy build: 5af87b4 

The bug is partially fixed, since the plot is not updated according to the loaded Scaling settings.