analogdevicesinc/scopy #417: oscilloscope acquisition modes

rgetz, 2018-04-26:
Most (all) digital oscilloscopes have a variety of different acquisition modes.

Acquisition modes are tuned sampling algorithms that give you unique insights into signals.  By varying the sample rate of the scope’s analog-to-digital converter (ADC) and selectively plotting or combining sample points, different characteristics of a signal can be observed.

This is described in a variety of different places:

http://www.electronicdesign.com/test-measurement/why-you-should-care-about-oscilloscope-acquisition-modes

https://www.testandmeasurementtips.com/the-oscilloscope%E2%80%99s-acquisition-modes/

http://anlage.umd.edu/Microwave%20Measurements%20for%20Personal%20Web%20Site/55W-17443-2.pdf

https://www.youtube.com/watch?v=_ai7dk0W73A

https://www.youtube.com/watch?v=86mGezxRSms

The most important one for many purposes, is the average mode, which saves a sample point taken during a waveform interval. The final displayed waveform is produced by time aligning the signal, and then averaging the waveform points from consecutive acquisitions (assumes a repeating signal). This mode minimizes noise without compromising bandwidth, and is very important when looking at constant state power supplies, since current probes are typically very noisy.

Couple papers which might help:

Alignment of Noisy Signals
https://ws680.nist.gov/publication/get_pdf.cfm?pub_id=901379

Effects of Averaging to Reject Unwanted Signals in Digital Sampling Oscilloscopes
http://investor.teradyne.com/File%20Library/Defense-Aero/Technical%20Papers/2010/Bishop_Effects-of-Averaging-to-Reject-Unwanted-Signals_2010.pdf

This seems to be somewhat related to #243

rgetz, 2018-04-27:
When displaying things in the time domain, alignment needs to happen (which isn't necessary in the frequency domain, since you aren't looking at phase). 

Without alignment, jitter due to the digital trigger will cause incorrect results at high frequency (large slew rate) signals.

damercer, 2018-04-27:
Robin is right to point out the differences in averaging when using time waveform data and averaging in the frequency domain. In a spectrum analyzer the point by point averaging is done on multiple amplitude/phase vs frequency arrays from multiple acquisitions. Point by point averaging across multiple amplitude vs time arrays needs to have a common trigger point across acquisitions, ie each acquisition needs to be shifted in time to align with trigger point. This may happen automatically if triggering is done in the hardware ahead of the data being sent to the GUI.

Doug