analogdevicesinc/scopy #420: Sometimes clicking connect/disconnect crashes Scopy

tagoylo, 2018-05-03:
Scopy build: [1aef8ae7](https://github.com/analogdevicesinc/scopy/commit/1aef8ae7093b852cb87adfa30ddd9b7734d2fd77)

When clicking connect/disconnect, Scopy sometimes crases. Though most of the time it recovers, there are still times that it crashes.

[image]

adisuciu, 2018-05-03:
What do you mean most of the time it recovers ? 
Also are there any other steps to reproduce this ?
We haven't experienced this here.

rgetz, 2018-05-03:
I think I have seen this if I double click / triple click things I shouldn't. It I treat it the way I should (single click, wait) - it works as expected. 

Once a connect (or disconnect) is clicked, is there a lock / mutes that stops another click from doing things simultaneously?

DanielGuramulta, 2018-05-03:
Fixed with commit: https://github.com/analogdevicesinc/scopy/commit/b4e53cb78bfdd04501539da3e0a24614c5a50afc

amiclaus, 2018-05-07:
On Scopy build: 5af87b4 the bug is fixed.

closed 2018-05-07 by amiclaus

rgetz, 2018-05-07:
Tony:

Normally we should let the person who reported the bug comment on the proposed fix before we close the issue. 

Trecia; did you look at this to confirm the fix?

tagoylo, 2018-05-07:
Hi Robin, 
Yes, the fix was confirmed. The bug doesn't exist in the latest build anymore.

amiclaus, 2018-05-07:
Robin:

This approach of waiting for the bug reporter to confirm the fix through a comment was not used before. That's why the steps of reproducing the bug are included in the description, so it can be also tested by others (I always test the fixed bug before closing it).

From now on, do you want me to wait the bug reporter to confirm the fix before testing/closing it?

Thank you