analogdevicesinc/scopy #423: +/- for horizontal position doesn't work

tagoylo, 2018-05-07:
Scopy build: [5af87b4e](https://github.com/analogdevicesinc/scopy/commit/5af87b4eb49eb576387e972cb58213495adc2252)

Upon launch, open oscilloscope.
Try clicking on +/- of horizontal position, it doesn't work on both channels.
It will work only after the user changes the value by entering in the widget or by moving the cursor.

[image]

AlexandraTrifan, 2018-05-07:
Fixed with commit a57d0f56ae663a422f6aaa3c716ae45e1c8a61f0.

tagoylo, 2018-05-09:
On Scopy build: [`70fb08da`](https://github.com/analogdevicesinc/scopy/commit/70fb08da741b42f0a0b985f4e1804f8bee2de8a1), the bug is fixed.

closed 2018-05-09 by tagoylo