analogdevicesinc/scopy #424: Spectrum Analyzer: Trace Types

drbautista, 2018-05-08:
The Spectrum Analyzer Features different trace type:

[image]

I'm currently not very familiar with all of these trace types but I can only get a response on Peak Hold Continuous and Min Hold Continuous mode. 
Are the other modes already implemented on Scopy? 
I compared it with Analog Discovery's Spectrum Analyzer. Analog Discovery's Spectrum Analyzer Trace type provide different response to the signal.

adisuciu, 2018-05-08:
What do you mean you only get response from PeakHold and MinHold.

All of them are implemented, you will get more diverse results if you put in a higher averaging value.

drbautista, 2018-05-09:
oh yeah, I think I may have missed changing the average value.
All of the trace types work Adrian. 

Thanks a lot. I'm closing the issue.

closed 2018-05-09 by drbautista