analogdevicesinc/scopy #433: Signal Generator: Loading waveform file crashes Scopy

drbautista, 2018-05-11:
OS: Windows 10 64-bit
 Scopy Version: 70be5dc
 Firmware Version: v0.19

Loading a .csv file and running the Signal Generator instrument crashes scopy.

claudiupop, 2018-05-11:
Is this happening with any .csv file or only with a specific file? If that's the case could you post the .csv file that crashes scopy?

drbautista, 2018-05-11:
I tried two .csv files which are designed to output a staircase signal.
One is the file I'm using on previous versions and the other is the one used on the BJT Curve tracer lab activity.
I attached it here.

[stairstepsignals.zip](https://github.com/analogdevicesinc/scopy/files/1994296/stairstepsignals.zip)

adisuciu, 2018-05-11:
Fixed with  ba343d6

adisuciu, 2018-05-11:
By the way, one of the fiels in  the archive is .xlsx which won't load properly or won't load at all in scopy.