analogdevicesinc/scopy #437: Binary Counter Frequency Issue

amiclaus, 2018-05-18:
Scopy build: 7d7f313

Group DIO0 - DIO3 and set the Binary Counter pattern.

Set Frequency to 50KHz. 
The output pattern is correct (4 bit counter):
[image]

Set Frequency to 51KHz.
The output is now corresponding to a 3-bit binary counter:
[image]

This issue is valid for multiple frequency intervals.

adisuciu, 2018-05-25:
Fixed with 36713226af6073ed0c3fb8e95689f19026d56544

amiclaus, 2018-07-31:
On Scopy build: 01d69df (v1.0.1) the bug is fixed.

closed 2018-07-31 by amiclaus