analogdevicesinc/scopy #454: Cannot connect remotely on Mac

amiclaus, 2018-07-13:
Scopy version v1.0

When trying to connect via IP 192.168.2.1 it fails.

HoRNDIS driver installed, but device not detected in the network.

amiclaus, 2018-07-13:
dmesg
https://pastebin.com/DZWss5Ke

mhennerich, 2018-07-13:
I assume this worked before? What's new then? The macOS version?  Scopy version doesn't matter... 
If you plug the same M2k into a windows box, does it work?

amiclaus, 2018-07-13:
On Windows, the same board is working.

mhennerich, 2018-07-13:
I don't have a mac here. Can someone in Cluj have a look?

amiclaus, 2018-07-13:
It has to do with the OS X version. On v10.11 (El Capitan), the connection in working. Probably it has to do with the driver on newer OS versions.

mhennerich, 2018-07-13:
@tfcollins probably knows how to fix this Sierra. The Pluto should have the same issue then.

tfcollins, 2018-07-13:
@amiclaus can you try HoRNDIS release 8.  Be sure to reboot before trying to talk to a device.

Here is the info from my mac:
```
Analogs-Mini:~ analog$ sw_vers
ProductName:    Mac OS X
ProductVersion: 10.13.2
BuildVersion:   17C88
```

```
Analogs-Mini:~ analog$ networksetup -getinfo "PlutoSDR (ADALM-PLUTO)"
DHCP Configuration
IP address: 192.168.2.10
Subnet mask: 255.255.255.0
Router: (null)
Client ID:
IPv6: Automatic
IPv6 IP address: none
IPv6 Router: none
Ethernet Address: 00:e0:22:85:61:95
```

```
Analogs-Mini:~ analog$ cd /Library/Frameworks/iio.framework/Tools/
Analogs-Mini:Tools analog$ ./iio_info -s
Library version: 0.15 (git tag: 6ecff5d)
Compiled with backends: xml ip usb
Available contexts:
        0: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=104473222a870017fdff070014ced4f484 [usb:20.3.5]
```

rgetz, 2018-07-14:
Can you not connect remotely? (Ie with a USB/Ethernet dongle?) or just RNDIS(Ethernet over USB?) 

Sounds like RNDIS, which is actually different than "remote".

tfcollins, 2018-07-14:
Both are functional
```
Analogs-Mini:Tools analog$ ./iio_attr -u ip:192.168.2.1 -C
IIO context with 8 attributes:
hw_model: Analog Devices PlutoSDR Rev.B (Z7010-AD9363)
hw_model_variant: 0
hw_serial: 104473222a870017fdff070014ced4f484
fw_version: v0.26
ad9361-phy,xo_correction: 39999881
ad9361-phy,model: ad9364
local,kernel: 4.9.0-10194-g40c2158
ip,ip-addr: 192.168.2.1
```

```
Analogs-Mini:Tools analog$ ./iio_attr -u usb:20.3.5 -C
IIO context with 13 attributes:
hw_model: Analog Devices PlutoSDR Rev.B (Z7010-AD9363)
hw_model_variant: 0
hw_serial: 104473222a870017fdff070014ced4f484
fw_version: v0.26
ad9361-phy,xo_correction: 39999881
ad9361-phy,model: ad9364
local,kernel: 4.9.0-10194-g40c2158
usb,idVendor: 0456
usb,idProduct: b673
usb,release: 2.0
usb,vendor: Analog Devices Inc.
usb,product: PlutoSDR (ADALM-PLUTO)
usb,serial: 104473222a870017fdff070014ced4f484
```

tfcollins, 2018-07-14:
I tried with Release 9 and the network context broke.

mikhailai, 2018-07-14:
Sure, I do ;). I did not know how y'all device looks on the USB bus, so probably missed that case. I'll promptly try to fix Rel 9 to support it, but I need more information. If anyone has access to a Linux box, can you please try to connect the device to it, set it up to use RNDIS, run "lsusb -v", and send me the output (limited to the given device, if you prefer - see lsusb options).

If you're only on Mac and using Macports, I believe you can get the "lsusb" by installing "usbutils". Unfortunately, I don't know the equvalent for Homebrew (which I'm using).

rgetz, 2018-07-14:
@mikhailai here you go. This is with a pre-production rev A with the latest firmware (v0.28) (which is all I have on me now). If you want/think it would help - I can send a device to you.

```
Bus 003 Device 015: ID 0456:b673 Analog Devices, Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0456 Analog Devices, Inc.
  idProduct          0xb673 
  bcdDevice            4.09
  iManufacturer           1 Analog Devices Inc.
  iProduct                2 PlutoSDR (ADALM-PLUTO)
  iSerial                 3 100000235523732200190024090216ed01
  bNumConfigurations      1
OTG Descriptor:
  bLength                 3
  bDescriptorType         9
  bmAttributes         0x03
    SRP (Session Request Protocol)
    HNP (Host Negotiation Protocol)
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          218
    bNumInterfaces          6
    bConfigurationValue     1
    iConfiguration          4 RNDIS/MSD/ACM/IIOUSBD
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       6 Ethernet Networking
      bFunctionProtocol       0 
      iFunction               7 RNDIS
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol    255 Vendor Specific (MSFT RNDIS?)
      iInterface              5 RNDIS Communications Control
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x00
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
[... 193 lines cut ...]
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0000
  (Bus Powered)
```

-Robin

mikhailai, 2018-07-16:
Thanks for the information, Robin. The RNDIS-related USB descriptors of PlutoSDR look very similar to another embedded Linux board that I have - PocketBeagle, so I think I got this covered now.
I now have the HoRNDIS build that should support PlutoSDR:
https://joshuawise.com/downloads/HoRNDIS-9.1d1.pkg
It should be of release quality, so please try it out and let me know if you see any issues. I want to make sure it covers this case, before pushing out the 9.1 release.

adisuciu, 2018-07-16:
Just tested with the pkg @mikhailai  attached and it works correcty.

Thanks !

rgetz, 2018-07-16:
@mikhailai thanks for your help, fix and quick pkg to test things out. 

@adisuciu - what version of the OS did you test things on - we have seen issues in the past with HoRNDIS that it would work in one version of OS-X, but not on a different one. I think @tfcollins can try different versions that you might have access to.

tfcollins, 2018-07-16:
Same here, all my macs are on 10.13.  High Sierra was released almost a year ago and Mojave (10.14) is close to release.  So older releases will be less common.

rgetz, 2018-07-16:
Ok - it was @amiclaus foo who said they had access to a 10.11 machine.

amiclaus, 2018-07-17:
@rgetz we have here a mac with multiple partitions. among them, there is one with High Sierra (10.13) and one with El Capitan (10.11).

amiclaus, 2018-07-17:
Just checked the RNDIS connection on 10.11 with the latest driver build and it works fine.

rgetz, 2018-07-17:
What we want to check compatibility with is the same versions that MATLAB supports, which is : 
macOS High Sierra (10.13)
macOS Sierra (10.12)
macOS El Capitan (10.11)

Per: https://www.mathworks.com/support/sysreq.html

So this can be closed now?

amiclaus, 2018-07-17:
I would suggest testing it also in Sierra (10.12), but we don't have it here. (Just to be 100% sure)

rgetz, 2018-07-17:
I don't think anyone has that.

rgetz, 2018-07-18:
so, I think this is Ok to be closed. @mikhailai thanks for your help/assistance.

I assume that 9.1 will show up at : https://github.com/jwise/HoRNDIS/releases ?

-Robin

closed 2018-07-18 by rgetz

mikhailai, 2018-07-23:
FYI: The HoRNDIS 9.1 was published.