analogdevicesinc/scopy #466: Cursor horizontal lock issue

tagoylo, 2018-07-31:
Scopy version: [01d69df8](https://github.com/analogdevicesinc/scopy/commit/01d69df8034b684733b406e720e467fd5ff2fcab)

Horizontal cursor moves on its own and messes with the lock configuration set by the user.

Steps to reproduce

Reset Scopy first
Enable Cursor
Move one of the horizontal cursors
Enable Horizontal cursor lock
Then Move horizontal cursors

As observed, the other horizontal cursor, which was not moved earlier, moves on its own and messes with the lock configuration set by the user.
[image]

DanielGuramulta, 2018-08-03:
Fixed with commit: https://github.com/analogdevicesinc/scopy/commit/850638f3556f4014f7c363d2e7298c01a6595edd

amiclaus, 2018-10-11:
On Scopy build (v1.0.2): d6b1429 the bug is fixed.

closed 2018-10-11 by amiclaus