analogdevicesinc/scopy #467: Channel output state issue after group pin session

tagoylo, 2018-07-31:
Scopy version: [01d69df8](https://github.com/analogdevicesinc/scopy/commit/01d69df8034b684733b406e720e467fd5ff2fcab)

Steps to reproduce

Set channels as Group pins.
Set DIO 0-7 as IN.
Set DIO 8-15 as OUT.
Change OUT value to 5 in DIO 8-15, then click Run.
The value should reflect on DIO 0-7.
Click Stop.
Set both Group as Individual pins.
Then click Run.
As seen in the picture below, the channel output state does not correspond to the actual output. It seems that in this case, the previous group output value, which is 5, is retained.

[image]

DanielGuramulta, 2018-08-03:
Fixed with commit: https://github.com/analogdevicesinc/scopy/commit/25579b94d15fba95a4d4a3ff7b41ea0549be965c

amiclaus, 2018-08-06:
On Scopy build: 7e560ab (v1.0.1) the bug is fixed.

closed 2018-08-06 by amiclaus