analogdevicesinc/scopy #472: Dots on cursor reappear

tagoylo, 2018-08-02:
Scopy version: [01d69df8](https://github.com/analogdevicesinc/scopy/commit/01d69df8034b684733b406e720e467fd5ff2fcab)

Steps to reproduce:

Run network analyzer.
Enable Cursor.
Then disable Cursor.
Dots and the cursor are now not displayed.
Then by clicking among any of the following, the dots reappear:

- general settings icon
- instrument settings icon
- logarithmic/linear
- Min Freq
- Max Freq
- Max Magnitude
- Max Phase

As observed the dots reappears on the plot.

[image]

DanielGuramulta, 2018-08-03:
Fixed with commit: https://github.com/analogdevicesinc/scopy/commit/c600154f8e926bdedaf0717335b27244c96b9685

amiclaus, 2018-08-06:
On Scopy build: 7e560ab (v1.0.1) the bug is fixed.

closed 2018-08-06 by amiclaus