analogdevicesinc/scopy #474: Inconsistent position of dots in phase when changing scale

tagoylo, 2018-08-02:
Scopy version: [01d69df8](https://github.com/analogdevicesinc/scopy/commit/01d69df8034b684733b406e720e467fd5ff2fcab)

Steps to reproduce:

Run network analyzer.
Enable Cursor.
Then change the scale to linear and vice versa.
As observed, the dots in phase does not stick with the cursor lines. 
It only gets back to the cursor lines when cursor setting is clicked.

[image]

DanielGuramulta, 2018-08-03:
Fixed with commit: https://github.com/analogdevicesinc/scopy/commit/536f6e3889c2de5d2a8550c3221b3bdcd4270f00

amiclaus, 2018-08-06:
On Scopy build: 7e560ab(v1.0.1) the bug is fixed.

closed 2018-08-06 by amiclaus