analogdevicesinc/scopy #475: Spectrum Analyzer: incorrect fundamental frequency reading

drbautista, 2018-08-02:
OS: Windows 10 64bit
 Scopy Version: c61d592
 Firmware Version: v0.20

Steps to Reproduce:

1.

dNechita, 2018-08-02:
You just experienced the aliasing effect.
Since you told the instrument to look for signals with frequencies up to 500 kHz, the sample rate has been automatically adjusted for this range of frequency (sample rate is now half of what it was when you first set the stop frequency to 1 MHz). As a consequence the 750 kHz signal and all signals with higher frequencies than 500 kHz cannot be sampled properly thus introducing alias components.
For more info lookup for aliasing and Nyquist.

rgetz, 2018-08-02:
Isn't the spectrum analyzer the max sample rate so this doesn't happen?

Why does the sample rate change?

adisuciu, 2018-08-03:
This is how I understand it: possible FFT bin sizes are fixed power of 2 values from 256 to 32768. In order to get as much of the signal as possible, we adjust the "oversampling_ratio" parameter to an appropriate value in order to get the samplerate at least 2x stop frequency.

If we keep samplerate at maximum value for all sweeps, we need to increase the buffer size by a lot for low stop frequencies, increasing both CPU and memory usage.

rgetz, 2018-08-03:
32k is realitivily small in terms of capture - isn't it? I get that this is directly relates to bin size and screen refresh rates. But what is settjng this limit? It should be able to go much higher if we wanted to. Shouldn't it? Or is there a limit in GNU Radio I don't understand?

You are right - it would be nice if one pixel = one bin, but users can already zoom in (set start/stop very close), that this breaks. We don't need less than one pixel per bin. 

What we want to do is have the sample rate as fast as possible to ensure aliasing doesn't occur, use that. What could happen is the do one FFT at max sample rate, analyze it, determine the max input frequency, and set the sample rate based on that. This doesn't need to display - only happen once every few seconds (or longer). 

It's common of spectrum analyzers to sweep, to not refresh the screen every 1/30 of a second - but update 1/5 or every second depending on the settings.

closed 2024-09-24 by adisuciu