analogdevicesinc/scopy #478: Signal shape modifies on Rise/Fall Time at specific frequencies

amiclaus, 2018-08-03:
Scopy version: e3b2b4e (v1.0.1)

Generate Square Wave at 200Hz and plot the signal's rise/fall time.

[image]

Set frequency to 400Hz.
[image]

Set frequency to 600Hz.

[image]

At any other frequency, the signal looks fine. (Picture below is at 201Hz)

[image]

adisuciu, 2018-08-03:
Should be fixed with https://github.com/analogdevicesinc/scopy/commit/a3ab408ce4106367c149404caf872bae77edfb8a

amiclaus, 2018-08-06:
Shot taken at 400Hz:
[image]

amiclaus, 2018-08-06:
Also, there is a significant difference in the rise/fall time i.e. between 200Hz and 201Hz.

200Hz
[image]

201Hz
[image]

adisuciu, 2019-02-27:
Fixed with 272b2454be569e85e99bd9b98ad029f0174d46be

amiclaus, 2019-05-09:
On Scopy version: e28686c (v1.0.5) the bug is fixed.

closed 2019-05-09 by amiclaus