analogdevicesinc/scopy #485: V 1.0.1 release version crashes on Windows 7 64 bit

damercer, 2018-08-17:
Hi:
I just (re)installed Scopy V1.0.1 (fresh download) from the Releases and after it runs and the opening screen appears, I select the M2k I have plugged into the USB port and then click on Connect. The following pops up and the program terminates.
[image]

Do I need a different runtime library now or something?

The M2k has latest firmware and runs fine with my Python programs (iio.py).

Thanks

Doug

adisuciu, 2018-08-17:
Can you share the .ini files ? It should be in C:\Users\<your user>\AppData\Roaming\ADI

damercer, 2018-08-19:
Attached is the contents of that directory:
[App Data.zip](https://github.com/analogdevicesinc/scopy/files/2300522/App.Data.zip)
As a test, I moved the files to another directory ( so that one is now empty ) and Scopy still crashes in the same way.
Doug

adisuciu, 2018-08-20:
We tried fixing it with 18f9fb0. I updated the installer in the release page. Can you download, install and make sure you restart your computer after installing, then get back to us and confirm that it works ? 

The new installer makes some modifications to the SCOPY_PYTHONPATH environment variable and I think a restart is required in order for changes to take effect.

damercer, 2018-08-20:
Scopy now connects and seems to operate. But the opening screen warns "Can't identify device. Please try to update your firmware!" As far as I can tell v0.20 of the firmware is the latest??

[image]

Why would it be saying I need newer firmware? Point me to newer firmware or remove this warning.

Doug

damercer, 2018-08-20:
By the way:
From the info.html file on this M2k:

Status of the M2k firmware:

Wow! Your M2k Firmware Version v0.20 is newer than (alpha - v0.20) on Github.

Maybe this is the problem?

Doug

adisuciu, 2018-08-20:
What hardware revision do you have ? 
On connect, scopy verifies that the m2k-fabric device has output voltage4 channel. 
https://github.com/analogdevicesinc/scopy/blob/master/src/info_page.cpp#L351

This channel overrides one of the LEDs on the board, in order to create the blinking-identify functionality. Your hardware might not have this functionality enabled, therefore you get the warning.

mhennerich, 2018-08-20:
The javascript which compares the versions on info.html can't handle the alpha tag.
The version you have installed is the latest.
From the screenshot posted it's HW Rev.B

I think Scopy should not ask to update the firmware in this case ...

adisuciu, 2018-08-20:
99% of the customers won't have rev. B, would they ? They will probably have some kind of firmware that doesn't recognize the LED voltage channel so updating the firmware would probably be the right way to go.

adisuciu, 2018-08-20:
We could put a message that says that your hardware revision doesn't support identify. That would be the correct approach in my opinion.

mhennerich, 2018-08-20:
Based on HW Rev. and Firmware Rev. we could give a more adequate call for action...

>Wow! Your M2k Firmware Version v0.20 is newer than (alpha - v0.20) on Github.

This is fixed now.

adisuciu, 2018-10-11:
Implemented with ed8376ad13f5f26a87df25a5bb9dcbde26a9d077

amiclaus, 2018-10-11:
Checked the issue with a Rev. B board and it's fixed. (Scopy build: d6b1429)

closed 2018-10-11 by amiclaus