analogdevicesinc/scopy #486: Poor Vertical resolution?

damercer, 2018-08-20:
Why is the vertical resolution so poor when Channel 2 is set to 20 mV/Div? Looks like the LSB is 14 mV. Souldn't it be about 10X better?

[image]

I'm trying to reproduce a new SHA experiment I developed for M1k on M2k. I'm trying to measure the hold mode pedestal step which is only about 5 mV.

Doug

adisuciu, 2018-08-20:
It looks like channel 2 is in low-gain mode, although it should be in high-gain. I'm investigating.

adisuciu, 2018-08-21:
It looks like you have a corner case where you set the position at -2.5V . 

This line sets the limits for low/high gain for the channels based on what can be viewed on the plot:
https://github.com/analogdevicesinc/scopy/blob/master/src/oscilloscope.cpp#L3918

We're still investigating, but this might not be a bug.

damercer, 2018-08-21:
It most certainly is a bug. You are not using the DC offset to it's 
fullest extent.

The hardware is capable of this combination because I do it ( and larger 
offsets ) using iiolib and Python.

Doug

damercer, 2018-08-30:
It doesn't look like you are using the hardware offset DAC AD5625?
The four channels in the AD5625 are used to offset the two AWG channels and two Scope channels.
Have you figured out how to use the HW offset functionality yet? Sorry if I seem surprised this major hardware feature has not been implemented.
After importing iio into Python and connecting to M2k I access the channels by:

```
# Offset DACs
ad5625 = ctx.find_device("ad5625")
AWG1Offset = ad5625.find_channel("voltage0", True)
AWG2Offset = ad5625.find_channel("voltage1", True)
Scope1Offset = ad5625.find_channel("voltage2", True)
Scope2Offset = ad5625.find_channel("voltage3", True)
```

Converting from requested voltage to raw binary is done much like the other paths in the hardware.

The amount of HW offset you can introduce is also a function of the input common mode voltage i.e. the voltage on the - scope input vs ground.

Doug

mbanciso, 2018-10-02:
will change to support +-5V offset on High Gain mode

amiclaus, 2018-10-11:
On Scopy build (v1.0.2): d6b1429 the bug is fixed.

closed 2018-10-11 by amiclaus