analogdevicesinc/scopy #490: Cursor line is uneven outside the plot area

tagoylo, 2018-09-10:
Scopy build: [27349d52](https://github.com/analogdevicesinc/scopy/commit/27349d52299c6a99cd33f4afdbb60dae5b81440f)

As seen below, the cursor line is uneven outside the plot area.

[image]

DanielGuramulta, 2018-09-10:
Do you have screen scaling enabled? If so, then it's the same issue as in here: https://github.com/analogdevicesinc/scopy/issues/473. If you don't have it enabled can you tell me what screen resolution your computer has because I am unable to reproduce the bug. Thank you!

tagoylo, 2018-09-10:
Hi, yes I have screen scaling enabled. Also verified if screen scaling is off, the issue is not showing.

adisuciu, 2018-09-12:
We cannot really guarantee that things won't look weird with Windows scaling enabled.

closed 2018-09-12 by adisuciu