analogdevicesinc/scopy #491: Oscilloscope sometimes triggers with one sample difference

adisuciu, 2018-09-12:
Steps to reproduce
- Feed in a high frequency signal (10MHz) to the oscilloscope
- Set timebase to 100ns/div
- Set trigger
- Observe that the signal doesn't always intersect with the trigger point

Could this be HDL related ?

lxTHU, 2018-12-09:
Like this compare of [ADALM2000 vs Analog Discovery 2 vs PicoScope 2205](https://www.eevblog.com/forum/testgear/adalm2000-vs-analog-discovery-2-vs-picoscope-2205-(2010my)/) shows: AD2 with Overs=16 and achieve 1.6GHz effective rate and without the trigger issue. Thank you a lot~

adisuciu, 2019-05-07:
Should be fixed with latest firmware version

amiclaus, 2019-05-09:
On Scopy version: e28686c (v1.0.5) with FW: v0.23 the bug is fixed.

closed 2019-05-09 by amiclaus