analogdevicesinc/scopy #492: To gray out data logging feature when turned off

tagoylo, 2018-10-08:
Scopy version: [1d173e63](https://github.com/analogdevicesinc/scopy/commit/1d173e633165f0632036963f0b15180598c5add9)
OS: Windows 10
Firmware version: 0.21

When data logging is turned off, the user can still configure the settings relating to data logging. this might cause confusion on the user. Perhaps graying out the feature just like the External Trigger settings in Oscilloscope.
[image]

AlexandraTrifan, 2018-10-10:
Commit d6b1429b7cd726317cac5fbf6e62d27d370e7f2b implements the following behaviour: The controls for data logging are greyed out when the Voltmeter is not running and the data logging is OFF. If we start the Voltmeter, the controls are not greyed out, because the user needs to be able to preset the data logging feature.

amiclaus, 2018-10-11:
On Scopy build (v1.0.2): d6b1429 the feature is implemented.

closed 2018-10-11 by amiclaus