analogdevicesinc/scopy #493: Spikes show in the oscilloscope reading of pattern generator output

tagoylo, 2018-10-08:
Scopy version: [1d173e63](https://github.com/analogdevicesinc/scopy/commit/1d173e633165f0632036963f0b15180598c5add9)
OS: WIndows 10
Firmware version: 0.21

There are spikes showing up when monitoring the pattern generator output on the ocilloscope. 
This happens when time base is set to 50 ms and below.

Steps to reproduce:
Enable CH0.
Set clock frequency to 5 KHz with the duty cycle of 50%. 
Run instrument.
Monitor CH0 through Oscilloscope.
Set time base to 50 ms or below.
The waveform looks like the screenshot below.
[image]

DanielGuramulta, 2018-10-08:
Should be fixed with commit: https://github.com/analogdevicesinc/scopy/commit/4c27bcc2da76d52d429599532a8486ba15e438b3

amiclaus, 2018-10-11:
On Scopy build (v1.0.2): d6b1429 the bug is fixed.

closed 2018-10-11 by amiclaus