analogdevicesinc/scopy #497: Trigger settings button in logic analyzer does not close/hide when clicked

tagoylo, 2018-10-09:
Scopy build: [4c27bcc2](https://github.com/analogdevicesinc/scopy/commit/4c27bcc2da76d52d429599532a8486ba15e438b3)
OS: Windows 10

The trigger settings button in logic analyzer does not close or hide when clicked.

Steps to reproduce:
Open logic analyzer.
Then click the Trigger settings button to open the panel.
Then click again on the button to close/hide it.
As observed, it does not close/hide when clicked. 

[image]

DanielGuramulta, 2018-10-09:
Should be fixed with commit: https://github.com/analogdevicesinc/scopy/commit/e3efb5903bd9b89aff8d275be25b43111a3c65b4

amiclaus, 2018-10-11:
On Scopy build (v1.0.2): d6b1429 the bug is fixed.

closed 2018-10-11 by amiclaus