analogdevicesinc/scopy #499: Print button is not uniform among instrument

tagoylo, 2018-10-09:
Scopy build: [e3efb590](https://github.com/analogdevicesinc/scopy/commit/e3efb5903bd9b89aff8d275be25b43111a3c65b4)

As seen in the picture below, the print button is not uniform among instrument
[image]

AlexandraTrifan, 2018-10-09:
Should be fixed with commit 27eebb880e1ca5b0f465163e9827a4fad6863917 . 
For the Logic Analyzer, since there is not enough space, the button has 2 states: The one you posted, and the following one, which is the same as in the Oscilloscope: 
[image]

amiclaus, 2018-10-11:
On Scopy build (v1.0.2): d6b1429 the bug is fixed.

closed 2018-10-11 by amiclaus