analogdevicesinc/scopy #500: Issue when detaching Logic Analyzer instrument

tagoylo, 2018-10-10:
Scopy Build: [27eebb88](https://github.com/analogdevicesinc/scopy/commit/27eebb880e1ca5b0f465163e9827a4fad6863917)
OS: Windows 10

When detaching logic analyzer from the main window, it shows this behavior that the detached window gets wider and wider and does not stop.

Step to reproduce:
Open scopy main window 
Detach Logic analyzer
The behavior should show immediately
This only happens on the logic analyzer instrument

[image]

AlexandraTrifan, 2018-10-10:
Should be fixed with 13e6bc942921201a77fed41821e7815bba2081fe

amiclaus, 2018-10-11:
On Scopy build (v1.0.2): d6b1429 the bug is fixed.

closed 2018-10-11 by amiclaus