analogdevicesinc/scopy #501: Animation in hiding other channel parameters in logic analyzer is not implemented

tagoylo, 2018-10-10:
Scopy Build: [27eebb88](https://github.com/analogdevicesinc/scopy/commit/27eebb880e1ca5b0f465163e9827a4fad6863917)
OS: Windows 10

Steps to reproduce:
Open preferences and Enable Animation
Open Logic Analyzer
Click on the arrow, boxed in the picture below 
When clicked there is no animation implemented although preferences have enabled the animation.

[image]

adisuciu, 2018-10-10:
We cannot fix this right now as it touches a lot of areas of the Logic Analyzer. We plan a major Logic Analyzer redesign, and we'll keep this in mind.

adisuciu, 2020-10-19:
This menu no longer exists in newest Scopy version.

closed 2020-10-19 by adisuciu