analogdevicesinc/scopy #502: Window size issue when detaching instrument

amiclaus, 2018-10-11:
Scopy version: d6b1429
OS: Windows

Select Oscilloscope instrument, then detach the instrument.
[image]

Maximize window and then restore to previous window size.
The default window size changes and it cannot me reduced to its initial size.
[image]

It comes back to initial size only after closing the detatched window, select the Osc instrument and detach it again.

DanielGuramulta, 2018-10-12:
Should be fixed with commit: https://github.com/analogdevicesinc/scopy/commit/4551dd3726f6ad3c11a2287d2e8592ba6e6ef422

amiclaus, 2018-10-12:
On Scopy build: 97405f9c (v1.2.0) the bug is fixed.

closed 2018-10-12 by amiclaus