analogdevicesinc/scopy #506: menu option for disabling the ADC decimation filters

mbanciso, 2018-11-07:
Implement a menu for enabling/disabling the decimation filters.

damercer, 2018-11-08:
I think there may need to be more than simply (current) filters / no filters ( i.e. simply dropping/skipping samples). The overshoot/undershoot is caused by the filters not being flat top response. There should be the option to do a linear averaging (box car filter) and maybe some other shape with a flat top like maybe a raised cosine?

The current filters were probably not a good choice for time domain scope display in the first place. (OK for SDR but not M2k)

IMHO

Doug

adisuciu, 2019-02-06:
Added option to remove HW filtering with 623cf33329fdec34c5428b71910bcb2745ae6cea . 
If we want to add additional filtering it should be added in hdl. TBD if we want to add this or not.

adisuciu, 2020-10-23:
I'll close this as it has been inactive for a while, nor did we talk about adding more filters. We can reopen if the need arises.

closed 2020-10-23 by adisuciu