analogdevicesinc/scopy #507: window10 or 7 32bit crash or freeze when try connect

anotech, 2018-11-08:
OS Version: Windows 10 (10.0)
FW Version: v0.22
SW Version: 2e7c098

Description of the bug:<description>
Steps to reproduce:
-
-

The ini files might be useful to reproduce the error.
The ini file is located at: C:/Users/auno/AppData/Roaming/ADI/Scopy.ini
Please consider attaching it.

AlexandraTrifan, 2018-11-08:
Hi, 
It looks like there is a small issue with the version of a library that's being used by Scopy for the 32 bit builds.
You can use the following version temporarily: https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/20140252/artifacts . Just download the scopy-setup.exe using the link provided above.
We will introduce a fix for this.

Thank you
-Alexandra

adisuciu, 2018-11-13:
Can you confirm that this version works for you ?

-Adrian

adisuciu, 2018-11-23:
Should be fixed with latest release

amiclaus, 2019-05-09:
On Scopy version: e28686c (v1.0.5) the bug is fixed.
Checked for both Win10 and Win7 32bit.

closed 2019-05-09 by amiclaus