analogdevicesinc/scopy #509: Network Analyzer - corrupted frequency scale

adisuciu, 2018-11-19:
https://ez.analog.com/university-program/f/q-a/103960/adalm2000---scopy-nwa---corrupted-frequency-scale

ReiRue
91.35.25.35
on Nov 17, 2018
test case:   impedance of a crystal

NWA settings:

   frequency range    4.15MHz - 4.23MHz

   ref. channel = 2,  1000 samples

   frequency axis:    two test cases     lin and log

Observations:

- linear frequency scale

   - center frequency

          could imagine 4.19MHz    ...    but not the strange 4.18429 MHz

   - width of frequency grid

          intervals are a mixture between 8 and 16kHz width

- log frequency scale

   - center frequency is missing

   - width of frequency grid

          intervals are a mixture between 8 and 16kHz width

Question:  why do you introduce a center frequency?

Best Regards     ReiRue

DanielGuramulta, 2019-05-07:
Bug fixed with pr: #571

amiclaus, 2019-05-09:
On Scopy version: e28686c (v1.0.5) the bug is fixed.

closed 2019-05-09 by amiclaus