analogdevicesinc/scopy #510: Logic analyzer: When changing Run Mode the plot becomes distorted

tagoylo, 2018-11-21:
Scopy build: [e54d2038](https://github.com/analogdevicesinc/scopy/commit/e54d2038ef09512594e1f8074d83d3368689cda4)
OS: Windows 7, Windows 10

Steps to reproduce:

Do not maximize the window.

Run pattern generator DIO0 as 5 KHz clock 
Run logic analyzer DIO0, trigger set to rising edge, Run mode - repeated
Then change run mode from repeated >> stream >> repeated
The distortion will show.

This does not show when window is maximized.

[image]

AlexandraTrifan, 2018-11-21:
Should be fixed with 91c4f0cb0367ebd8d5f2e26ea69c5ada9f2210b8.

tagoylo, 2019-05-10:
On Scopy version f2eb3883 (v1.0.5), the bug is fixed.

closed 2019-05-10 by tagoylo