analogdevicesinc/scopy #511: Logic Analyzer: Enabled DIOs' Outputs are not working

HRosete, 2018-11-22:
Scopy build: e54d2038
OS: Windows 7, Windows 10

Steps to reproduce:

Run Pattern Generator. Enable DIO0-DIO7 then group them.
Set Pattern to "Binary Counter" and Frequency to 5kHz.
Run Logic Analyzer. Select DIO0-DIO7.
Monitor outputs.

Issue:
DIO0's and DIO1's outputs are up and working. However, DIO2-DIO7 are not.
[image]

adisuciu, 2018-11-22:
We rolled back libiio to v0.15 with 8fd35c1  
We rebuilt scopy with the latest dependencies and it should now work correctly.
https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/20491135

amiclaus, 2019-05-09:
On Scopy version: e28686c (v1.0.5) the bug is fixed.

closed 2019-05-09 by amiclaus