analogdevicesinc/scopy #513: Signal Generator - distortion of signal level

adisuciu, 2018-11-27:
https://ez.analog.com/university-program/f/q-a/104296/adalm2000---scopy---signal-generator---distortion-of-signal-level

 ReiRue
on Nov 26, 2018

I was checking the relative frequency resolution of the Signal Generator. Resolution for the tested frequencies was better than 1ppm.

-> excellent for the analysis of narrowband systems!

By accident I observed a strange signal level distortion. Test setup for the Signal Generator:
sine, 4V, frequency 20kHz   observed by Scopy Oscilloscope and external oscilloscope
(Scopy v1.0.2, FW v0.20, Windows 7)

Scopy Oscilloscope observed two discrete signal levels (no visible sine distortion!)

Vpp 3.94V and Vpp 3.26V   -> 17% amplitude drop

let's call 3.94V "H" and 3.26V "L", frequencies are in kHz

20.000 H,  20.001 L,  20.002 L,  20.003 L,  20.004 H,

20.005 H,  20.006 H,  20.007 H,  20.008 H,  20.009 L,

20.010 H,  20.011 L,  20.012 H,  20.013 H,  20.014 L,

20.015 H,  20.016 H,  20.017 L,  20.018 H  ,20.019 L,

...

please check with the above frequencies (most other frequencies are clean)!

Potential impact:

- glitches in frequency sweeps (e.g. Network Analyzer)

Best Regards      ReiRue

DanielGuramulta, 2019-05-07:
Should be fixed with pr: #580

amiclaus, 2019-05-09:
On Scopy version: e28686c (v1.0.5) the bug is fixed.

closed 2019-05-09 by amiclaus