analogdevicesinc/scopy #516: Phase range enhancement

amiclaus, 2018-12-03:
Currently the maximum range supported by NA is between -180deg and 180 deg

For better data visualization in certain circumstances and avoid the phase jumps between plot edges, the user should be able to adjust the phase display as follows:

Min 0deg -> Max 360deg
Min -360deg -> Max 0deg

damercer, 2018-12-10:
I don't know if this is allowed but is it possible to have the phase plot scale be smaller than a full 360 degrees (or -180to+180)? A way to zoom in as it were if the phase only changes 90 degrees over the sweep.

Doug

adisuciu, 2018-12-11:
Yes, we'll take a look at that.

DanielGuramulta, 2019-05-07:
Feature implemented with pr: #571

amiclaus, 2019-05-09:
On Scopy version: e28686c (v1.0.5) the feature is implemented.

closed 2019-05-09 by amiclaus