analogdevicesinc/scopy #520: Pattern Generator / Logic Analizer support in scripting

easyvolts, 2018-12-13:
I'm not sure if it's the right place to suggest an enhancement to the Scopy, please correct me if needed. My suggestion is to add in the Scopy special commands to control Pattern generator and Logic analyzer from scripts (probably the commands should be similar to Linux ones), to let scripts interact with external hardware by means of standard interfaces (like I2C, SPI, UART, etc). It would add a huge value to the ADALM2000 device and make it a more flexible link between PC and the real world. Thanks!

adisuciu, 2019-01-07:
Thanks for the suggestion, we'll look into it !

adisuciu, 2019-07-10:
Tracked here https://github.com/analogdevicesinc/libm2k/issues/4

closed 2019-07-10 by adisuciu