analogdevicesinc/scopy #522: Export to VCD - several bugs

gotchafr, 2019-01-16:
Hi there!

I've been playing with the M2K's logic analyzer and found that as it, VCD exports are broken. The issue here is that files exported in this format cannot be imported by tools like Pulseview or GTKView.

Looking at the file and the code I found these problems in the function `bool LogicAnalyzer::exportVCD`:

1. Timestamps in float -> AFAIK they should be integers (IEEE Verilog HDL 1364-2001)

    `timestamp = ((i * active_plot_timebase * 10) / sample_count)/timescale;`

2. Missing space between wire count and wire char id in vcd header signal declaration

    ```
    out << startSep << "var wire 1" << c << " DIO" <<
                                  |__ here
3. Export uses display (plot) timebase settings for  vcd header instead of actual sampling interval of data
    ```
    if( active_plot_timebase < 1e-6 ) {
    		timescaleFormat = "ns";
    		timescale = 1e-9;
    }
4. In the channel loop for dumping data, channel counter `p` is not incremented if no change is detected

    ```
    if((current_bit == prev_bit) && (i != 0))
                 continue;` 

Here is a debugged working version of the export function

 ````

    bool LogicAnalyzer::exportVCD(QString filename, QString startSep, QString endSep)
    {
    	uint64_t current_sample, prev_sample;
    	int current_bit, prev_bit, p;
    	QString timescaleFormat;
    	double timescale;
    	double timestamp;
    	bool timestamp_written = false;
    
    if( active_sampleRate == 0 ) {
                return false;
    
[ExportVCD_partial_patch.txt](https://github.com/analogdevicesinc/scopy/files/2763523/ExportVCD_partial_patch.txt)
[ExportVCD_partial_patch.txt](https://github.com/analogdevicesinc/scopy/files/2763525/ExportVCD_partial_patch.txt)

}

	QFile file(filename);
	if( !file.open(QIODevice::Append)) {
		return false;
	}

	QTextStream out(&file);

	/* Write the specific header */
	timescale = 1 / active_sampleRate;
	if( timescale < 1e-6 ) {
		timescaleFormat = "ns";
		timescale *= 1e9;
	}
	else if( timescale < 1e-3 ) {
		timescaleFormat = "us";
		timescale *= 1e6;
	}
	else if( timescale < 1 ) {
		timescaleFormat = "ms";
		timescale *= 1e3;
	}
	else
		timescaleFormat = "s";

	out << startSep << "timescale " << QString::number(timescale) << " " << timescaleFormat << endSep;
	out << startSep << "scope module Scopy" << endSep;
	int counter = 0;
	for(int ch = 0; ch < no_channels; ch++) {
		if( exportConfig[ch] ) {
			char c = '!' + counter;
			out << startSep << "var wire 1 " << c << " DIO" <<
			       QString::number(ch) << endSep;
			counter++;
		}
	}
	out << startSep << "upscope" << endSep;
	out << startSep << "enddefinitions" << endSep;

	/* Write the values */
	std::shared_ptr<pv::data::Logic> logic_data = main_win->session_.get_logic_data();
	if(logic_data) {
		shared_ptr<pv::data::LogicSegment> segment = logic_data->logic_segments().front();
		uint64_t sample_count = segment->get_sample_count();
		for(uint64_t i = 0; i < sample_count; i++) {
			current_sample = segment->get_sample(i);
			if( i == 0)
				prev_sample = current_sample;
			else
				prev_sample = segment->get_sample(i-1);
			timestamp_written = false;
			p = 0;
			timestamp = ((i * active_plot_timebase * 10) / sample_count)/timescale;

			for(int ch = 0; ch < no_channels; ch++) {
				current_bit = (current_sample >> ch) & 1;
				prev_bit = (prev_sample >> ch) & 1;

				if((current_bit == prev_bit) && (i != 0)) {
					p++;
					continue;
				}
				if( !timestamp_written )
					out << "#" << QString::number(i);
				if(exportConfig[ch]) {
					char c = '0' + current_bit;
					char c2 = '!' + p;
					out << ' ' << c << c2;
					p++;
					timestamp_written = true;
				}
			}
			if(timestamp_written)
				out << "\n";
		}
	}
	else {
		file.close();
		return false;
	}
	file.close();
	return true;
}

AlexandraTrifan, 2019-01-16:
Hi, 

Thank you very much for your feedback and your work on this. 
Would you like to group your fixes into a commit, sign it and create a pull request? This way we can review and merge your work to the master branch.

Thank you.
-Alexandra

gotchafr, 2019-01-16:
Hi Alexandra,
You are welcome... I did this fix and thought it could be useful for everybody...
BTW, Travis seems to have some trouble to build the project, is it misconfigured or something?
Cheers.

AlexandraTrifan, 2019-01-16:
Thank you for you PR, we'll review and merge it soon. 
The Travis scripts are work in progress at the moment. So, right now, it's not an issue if the PR Travis check fails.

-Alexandra

amiclaus, 2019-05-09:
On Scopy version: e28686c (v1.0.5) the bug is fixed.

closed 2019-05-09 by amiclaus