analogdevicesinc/scopy #524: Logic Analyzer - Export to CSV - extra comma at the end of the line

gotchafr, 2019-01-17:


AlexandraTrifan, 2019-01-17:
Hi,

You can open a Pull Request for this as well, in order for us to review it and merge it into master.
On your forked repository, create a new branch for this fix, create a commit and sign it using --signoff in the git commit command (for example: git commit -m "You commit message" --signoff ) . This will add a line in your commit message, similar to the following: Signed-off-by: Your name <youremail@email.com> . 
 
Please let us know if you have any questions.
Thank you

-Alexandra

gotchafr, 2019-01-18:
Hi,
should I branch on my last commit or on the v1.0.3 commit?
Cheers.

AlexandraTrifan, 2019-01-21:
Hi,
You can use the latest analogdevicesinc/scopy master branch.

-Alexandra

amiclaus, 2019-05-09:
Nitpick:
On Scopy build: e28686c (v1.0.5)
The header of the table: `Sample,Channel 0,Channel 1,` still contains an extra comma at the end of the line.
Should be a quick fix @AlexandraTrifan

adisuciu, 2019-05-14:
https://github.com/analogdevicesinc/scopy/pull/586

fixed

gotchafr, 2019-05-14:
Hi Adrian,
I just looked at your commit... I think there is a misunderstanding here:
The problem is the extra separators at the end of the line... And if I understand correctly you are now just skipping the first one.
Cheers.

adisuciu, 2019-05-14:
Well, yes, but now I'm also adding the separator before the value.

amiclaus, 2019-05-14:
@gotchafr you can check if the exported csv file corresponds to the standard by downloading the latest build: https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/24523906/job/fych5dkfvuk9xrt8/artifacts

If everything works fine, please let us know so we can close the bug.

gotchafr, 2019-05-14:
Yes, of course... I should have looked better: the name of the flag tricked me.
Cheers!

closed 2019-05-14 by amiclaus