analogdevicesinc/scopy #526: spectral measurements (noise floor, SFDR, SNR, THD, etc.)

janoschsimon, 2019-01-24:
hey all today i got the ADALM2000 and based on that page: https://wiki.analog.com/university/tools/m2k
i bought it to measure THD i downloaded the latest version on scopy but there is nothing to calculate the THD?

also Audio related a sine sweep would be nice from 20hz to 20khz :)
cheers and thx janosch

rgetz, 2019-01-24:
Thanks for the feedback. 

What sort of spectral measurements are you interested in. (THD, SNR, SFDR?)

Network Analyzer should work to 20kHz. (But be slow).

janoschsimon, 2019-01-24:
hmmm THD and THD+N would be nice :) actually all 
oh in the Network Analyzer tab i ca get a THD measurement? HMMM i was just looking in the Spectrum analyser tab.
But why all the spectral measurements are Advertised? or does it just mean that the ADALM2000 would be capable of those measurements? :)

damercer, 2019-01-28:
Just a quick observation about using the M2000 for Audio testing. Measurements like THD and noise on Audio signals generally need 16 bits or better. The 12 bit ADC/DAC in the M2000 will not be the best choice for this kind of measurements. The very high high potential sample rate ( 100MSPS ) does help the possible dynamic range by using a huge amount of sample averaging (decimation filtering) but only on the noise floor not the linearity (THD) on the ADC input channels. On the DAC output as an Audio source you are not going to be very happy with the results, again for Audio tests.

I would recommend using the M1000 with its 16 bit ADC ( at 100 KSPS ) for Audio testing.

But neither will do as well as a high end PC sound card and the appropriate software ( or even a garden variety sound card ) .

Just my 2 cents.

Doug

janoschsimon, 2019-01-28:
hey doug i came to the same conclusion :D but for things like impulse response it should work fine or? :)
also a sine sweep from 20hz to 20000hz would be very handy :)

cheers and the janosch

damercer, 2019-01-28:
For designing and testing audio filters (like impulse response) and making Bode blots, using the the M2000 and the Scopy Network Analyzer tool will work, but it's bandwidth is perhaps overkill for 20 Hz to 20 KHz. Also the M1000 would work equally well over that frequency range.
About the only advantage of the ALM hardware over a PC sound card is if response down to DC is of any interest. Sound cards generally have a DC blocking high pass response below 20 Hz or so.

Doug

janoschsimon, 2019-01-30:
the m2000 i already got :)
so where can i ask for the feature of the wavegenerator to output a sine sweep? :)

adisuciu, 2019-01-30:
We'll look into it. Until then you can probably create a chirp using python(for example) in a CSV file and load it into the signal generator.
-Adrian

damercer, 2019-01-30:
Hi @janoschsimon 

You could use another Audio processing program like Audacity which can generate a chirp (frequency sweep) waveform rather than writing Python code. You might need to convert the .wav format file to .csv if Scopy can't read .wav directly.
The only issue I might see is the length of the waveform needed (i.e. number of samples) might be longer than the maximum buffer size allowed in Scopy. To get reasonable frequency resolution for a full sweep from 20 Hz to 20 KHz you might need 1 second worth of data samples or more at around 48 KSPS at least. You will probably also need to set Audacity to generate the chirp at one of Scopy/M2000's fixed sample rates.

Doug

adisuciu, 2019-01-30:
Scopy should be able to handle more than 50k samples. It should also be able to handle WAV files directly if they are in 16bit integer format (the default format). After loading you can even change the sample rate & amplitude of the signal.
-Adrian

janoschsimon, 2019-01-30:
i will try that this evening thx for the tips :)

ferret-guy, 2020-03-10:
I wanted to chime in and say that I would be very interested in THD(+N), and SFDR measurements. I am currently making these in python from imported data, it's a pretty slow process. It is plenty good to test simple transistor amplifiers or 8-bit DACs.

rgetz, 2020-03-10:
@ferret-guy : is SINAD, SNR, THD, THD + N, SFDR, and Noise Floor the measurements you are interested in? I think those are the ones that we are looking at (all single tone measurements).

https://www.analog.com/media/en/training-seminars/tutorials/MT-003.pdf

We were not planning on looking at any two tone Intermodulation Products.

https://www.analog.com/media/en/training-seminars/tutorials/MT-012.pdf

ferret-guy, 2020-03-10:
@rgetz Yep, those would be great!

Intermodulation or third-order intercept would be nice though I assume that only a small fraction of the people who do single tone measurements would be interested in multi-tone measurements.

ferret-guy, 2021-02-16:
I wanted to ping this issue again to mention that I am still interested in this.

I would be willing to implement the code for the readings, but right now there are no global stats for the spectrum analyzer. That would require some infrastructure work.

adisuciu, 2021-02-17:
We have some of this feature implemented, however it does not fit currently with our development plan for this release. We will revisit this next release (end of the year 2021) - https://github.com/analogdevicesinc/scopy/pull/979

rgetz, 2021-06-30:
I think we have genanalyzer - a library we are working on - to do all the measurements - it just needs to be integrated.