analogdevicesinc/scopy #528: ADALM2000: Starts with a large DC offsets

acdcasic, 2019-02-04:
OS Version: Windows 7 SP 1 (6.1)
FW Version: v0.22
SW Version: 91c4f0c

Description of the bug:<description>
Steps to reproduce:
- Start Scopy Oscilloscope (Hit "Run" button)
- Ch1 and 2 shows approx. 2.5V DC offset (both must be 0V)
- Stop once and restart by "Run" button, then Ch1&2 show correct levels

See attached screenshot video
[AGDRec_20190204_165716.zip](https://github.com/analogdevicesinc/scopy/files/2826991/AGDRec_20190204_165716.zip)

[Scopy.zip](https://github.com/analogdevicesinc/scopy/files/2826995/Scopy.zip)

The ini files might be useful to reproduce the error.
The ini file is located at: C:/Users/mendo/AppData/Roaming/ADI/Scopy.ini
Please consider attaching it.

AlexandraTrifan, 2019-02-04:
This is fixed with the following commit: 855c6d15ee8777d1f8be45c766077fb196f2433d .

amiclaus, 2019-05-08:
On Scopy version: e28686c (v1.0.5) the bug is fixed.

closed 2019-05-08 by amiclaus