analogdevicesinc/scopy #529: Plot curve visualization options

adisuciu, 2019-02-04:
QwtPlot supports multiple types of data rendering. Currently we only render data using lines (or interpolation method). Implement other visualization methods (like point or sample/hold)

damercer, 2019-02-05:
I think in Scopy with the high sample rate of M2K and the limit set on the smallest time/div setting allowed, the sample points are closely spaced so that drawing straight lines between points is often good enough to render nice traces. However another rendering option you might look at is some sort of smooth curves like splines to connect the dots. Again, this only helps visually if the sample points are many pixels apart ( like having only a few samples per horizontal time division ).
Anyway, just a thought.
Doug

adisuciu, 2019-04-04:
Fixed with https://github.com/analogdevicesinc/scopy/pull/556

amiclaus, 2019-05-08:
On Scopy version: e28686c (v1.0.5) the feature is implemented

closed 2019-05-08 by amiclaus