analogdevicesinc/scopy #532: ADALM2000: SW crash on hardware connect

vthielen, 2019-02-13:
OS Version: Windows 7 SP 1 (6.1)
FW Version: 
SW Version: 91c4f0c

Description of the bug:
After installing drivers, upgrading firmware (now V0.22, : IIO version 0.15) and connecting hardware for first time:
- When I start the program, the main window appears. 
- When I try to connect to the device, windows says program has stopped.
- When I try to launch program again (several times), nothing happens.
- When I unplug the USB of the device and plug it in to a different USB port. Then, sometimes I can re-start the software, and I can connect and start instruments normally.

Devices in windows device manager are the same before and after the crash:
[image]

The ini file:

[DMMSpinBoxTimer15]
fine_mode=false

[EmptyPatternUISpinBoxDuty7]
fine_mode=false

[EmptyPatternUISpinBoxFrequency5]
fine_mode=false

[EmptyPatternUISpinBoxPhase6]
fine_mode=false

[LogicAnalyzerSpinBoxFrequency4]
fine_mode=false

[LogicAnalyzerSpinBoxPosition3]
fine_mode=false

[LogicAnalyzerSpinBoxTimeBase2]
fine_mode=false

[NetworkAnalyzerSpinBoxAmplitude26]
fine_mode=false

[NetworkAnalyzerSpinBoxMax.Magnitude28]
fine_mode=false

[NetworkAnalyzerSpinBoxMax.Phase30]
fine_mode=false

[NetworkAnalyzerSpinBoxMaxFreq25]
fine_mode=false

[NetworkAnalyzerSpinBoxMin.Magnitude29]
fine_mode=false

[NetworkAnalyzerSpinBoxMin.Phase31]
fine_mode=false

[NetworkAnalyzerSpinBoxMinFreq24]
fine_mode=false

[NetworkAnalyzerSpinBoxOffset27]
fine_mode=false

[NetworkAnalyzerSpinBoxSamplescount23]
fine_mode=false

[OscilloscopeSpinBoxPosition12]
fine_mode=false

[OscilloscopeSpinBoxPosition14]
fine_mode=false

[... 911 lines cut ...]
markers\6\freq=0
markers\7\chId=1
markers\7\mkId=1
markers\7\en=false
markers\7\visible=false
markers\7\type=-1
markers\7\freq=0
markers\8\chId=1
markers\8\mkId=2
markers\8\en=false
markers\8\visible=false
markers\8\type=-1
markers\8\freq=0
markers\9\chId=1
markers\9\mkId=3
markers\9\en=false
markers\9\visible=false
markers\9\type=-1
markers\9\freq=0
markers\10\chId=1
markers\10\mkId=4
markers\10\en=false
markers\10\visible=false
markers\10\type=-1
markers\10\freq=0

adisuciu, 2019-02-13:
Hey,

To me it sounds like there is a faulty connection to the computer. It can be either the USB cable or the USB port.
Can you try using a different USB cable ? If that doesn't work, can you try powering the board from an external source like a USB power brick ? (you should use the side USB port on the M2K for power while using the central USB for data - connected to the computer)

-Adrian

vthielen, 2019-02-13:
I tried it with another cable and a power brick, and it stays the same.

My first message was misleading, because the ADALM2000 unit doesn't seem involved:
Even if it's not connected, the software sometimes starts, and sometimes doesn't (window doesn't even appear).
When it appears, it stays open as long as I don't do anything, and crashes when I start to move the mouse cursor over it, even without clicking on anything.

vthielen, 2019-02-13:
I had a look in windows the event viewer, and found the information below about the crash (unfortunately in dutch):

Foutbucket 2600595602, type 473729084
Naam van gebeurtenis: APPCRASH
Antwoord: Niet beschikbaar
Id van CAB-bestand: 0

Handtekening van probleem:
P1: Scopy.exe
P2: 1.0.3.0
P3: 5bf7e5fc
P4: libusb-1.0.dll
P5: 1.0.21.11156
P6: 00000000
P7: c0000005
P8: 000000000000ac77
P9: 
P10: 

Bijgevoegde bestanden:
C:\Users\bart.vanthielen\AppData\Local\Temp\WER323C.tmp.WERInternalMetadata.xml

Deze bestanden zijn mogelijk hier beschikbaar:
C:\Users\bart.vanthielen\AppData\Local\Microsoft\Windows\WER\ReportArchive\AppCrash_Scopy.exe_18a2bdd7e4851be9326609d31bd77b606616d2_1c144973

The file mentioned above has following content:
Version=1
EventType=APPCRASH
EventTime=131945432594797804
ReportType=2
Consent=1
UploadTime=131945432596197818
ReportIdentifier=3d22a840-2f9f-11e9-b99f-00e02247042a
IntegratorReportIdentifier=3d22a83f-2f9f-11e9-b99f-00e02247042a
Response.BucketId=2600595602
Response.BucketTable=473729084
Response.type=4
Sig[0].Name=Naam van de toepassing
Sig[0].Value=Scopy.exe
Sig[1].Name=Versie van toepassing
Sig[1].Value=1.0.3.0
Sig[2].Name=Tijdstempel van toepassing
Sig[2].Value=5bf7e5fc
Sig[3].Name=Naam van foutmodule
Sig[3].Value=libusb-1.0.dll
Sig[4].Name=Versie van foutmodule
Sig[4].Value=1.0.21.11156
Sig[5].Name=Tijdstempel van foutmodule
Sig[5].Value=00000000
Sig[6].Name=Uitzonderingscode
Sig[6].Value=c0000005
Sig[7].Name=Uitzonderingsmarge
Sig[7].Value=000000000000ac77
DynamicSig[1].Name=Versie van besturingssysteem
DynamicSig[1].Value=6.1.7601.2.1.0.256.48
DynamicSig[2].Name=Landinstelling-id
DynamicSig[2].Value=2067
DynamicSig[22].Name=Aanvullende informatie 1
DynamicSig[22].Value=5ea2
DynamicSig[23].Name=Aanvullende informatie 2
DynamicSig[23].Value=5ea2f135d5f72b66a949a603400ead38
DynamicSig[24].Name=Aanvullende informatie 3
DynamicSig[24].Value=f956
DynamicSig[25].Name=Aanvullende informatie 4
DynamicSig[25].Value=f9560cf1d8040a6fa6c035b2b869643c
UI[2]=C:\Program Files\Scopy\Scopy.exe
UI[3]=Scopy - A software oscilloscope and signal analysis toolset werkt niet meer
UI[4]=U kunt online naar een oplossing voor het probleem zoeken.
UI[5]=Online naar een oplossing zoeken en dit programma sluiten
UI[6]=Later online naar een oplossing zoeken en dit programma sluiten
[... 116 lines cut ...]
LoadedModule[115]=C:\Program Files\Scopy\platforms\qwindows.dll
LoadedModule[116]=C:\Program Files\NVIDIA Corporation\nview\nViewH64.dll
LoadedModule[117]=C:\Windows\system32\ntmarta.dll
LoadedModule[118]=C:\Windows\system32\WLDAP32.dll
LoadedModule[119]=C:\Program Files\Scopy\imageformats\qgif.dll
LoadedModule[120]=C:\Program Files\Scopy\imageformats\qicns.dll
LoadedModule[121]=C:\Program Files\Scopy\imageformats\qico.dll
LoadedModule[122]=C:\Program Files\Scopy\imageformats\qjpeg.dll
LoadedModule[123]=C:\Program Files\Scopy\libjpeg-8.dll
LoadedModule[124]=C:\Program Files\Scopy\imageformats\qsvg.dll
LoadedModule[125]=C:\Program Files\Scopy\imageformats\qtga.dll
LoadedModule[126]=C:\Program Files\Scopy\imageformats\qwbmp.dll
LoadedModule[127]=C:\Program Files\Scopy\iconengines\qsvgicon.dll
LoadedModule[128]=C:\Windows\system32\WINTRUST.dll
LoadedModule[129]=C:\Windows\system32\RpcRtRemote.dll
LoadedModule[130]=C:\Windows\system32\WinUSB.DLL
LoadedModule[131]=C:\Windows\system32\hid.DLL
State[0].Key=Transport.DoneStage1
State[0].Value=1
State[1].Key=DataRequest
State[1].Value=Bucket=-1694371694/nBucketTable=473729084/nResponse=1/n
FriendlyEventName=Werkt niet meer
ConsentKey=APPCRASH
AppName=Scopy - A software oscilloscope and signal analysis toolset
AppPath=C:\Program Files\Scopy\Scopy.exe

adisuciu, 2019-02-13:
We're looking into this .. 

Can you share your computer specs, laptop brand etc. as well as if you are using a docking station for connection (we've had issues with those in the past)

-Adrian

vthielen, 2019-02-13:
The computer is an HP Z400 desktop, running win 7 professional SP1, having 6GB RAM 
Processor is Intel Xeon W3520, 64 Bit
I am not using a docking station.

The crash report above is with the ADALM2000 disconnected.
I saw in the report that it might have something to do with libusb, but I didn't try to connect. Does the program try to connect automatically when it starts up?

adisuciu, 2019-02-13:
Yes, It scans the USB every 5 seconds for IIO devices - https://github.com/analogdevicesinc/scopy/blob/master/src/tool_launcher.cpp#L513

So when you plug a device in, it automatically is shown in the launcher
We haven't had issues with that in the past though.

-Adrian

vthielen, 2019-02-14:
I changed my registry to make the program create a coredump file on crash.
Unfortunately, I cannot attach the file because it is too big (72MB) can I mail it to you?

Bart

adisuciu, 2019-02-14:
Hi,

Can you upload it to a file hosting site like transfer.sh and provide the link ?
Thanks! 

-Adrian

vthielen, 2019-02-14:
maybe you can take it from my Google drive:
https://drive.google.com/file/d/1y5eZ5zgKIR81oXC2Ry7XeG3IBoO-Y7LU/view?usp=sharing

adisuciu, 2019-02-18:
Hi,

Thanks for the coredump. The software crashes on trying to find USB devices. We still cannot reproduce the issue, it might have to do with the PCs USB controller or the Windows USB drivers. We're still looking into this.

-Adrian

pantechsolutions, 2019-06-26:
did u solve the issue?i am getting same kind of issue in windows 7

vthielen, 2019-06-26:
Hi,

No,
I contacted the people responsible for the software, but I didn't hear from
them any more.

Kind regards,

Bart Van Thielen

pantechsolutions, 2019-06-26:
Thank you for your reply.

elektrinis, 2019-10-09:
Hi,

I have a similar issue: Scopy crashes when trying to connect to hardware.
Scopy release v1.1.1 and today's nighty build
ADALM2000 fw. v0.22 and v0.24
PC: Desktop (i7-8700, 24GB RAM), Windows 10 x64 and Laptop with similar specs and OS (yes, happening on two machines)

I launch Scopy, it shows ADALM2000 in the list. I select it and click Connect. In that moment it crashes.
Tried so far without success:
* Relaunch Scopy.
* Plug HW in when scopy is launched.
* Reinstall Scopy.
* Reinstall driver.
* Delete AppData Roaming folder
* Use different USB cable and port
* Update Windows to latest feature pack

closed 2020-08-12 by adisuciu