analogdevicesinc/scopy #534: Scopy crashes on Mac

jcristine, 2019-02-16:
MacOS / Scopy Version - 91c4f0c.  Crashes on connect to device -- every time. 

Running on MacOS 10.14.2 -- MacBook Pro (Retina, 13-inch, Late 2012).  

I've tried both USB and TCP/IP both see the device and its context but clicking "connect" AFTER successfully adding the device causes a crash - EVERY time.

I was able to get the device to work under Windows 10 - first try, only using TCP/IP, no problems on Win 10 but no luck on the Mac.

When I run at the MacOS terminal I see this:

Warning: Using function expressions as statements in scripts in not compliant with the ECMAScript specification at
"function(o) { for (each in o) { print(each); } }..."
This will throw a syntax error in Qt 5.12. If you want a function expression, surround it by parentheses.
Illegal instruction: 4

This is from lldb:

Process 7965 stopped

* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_INSTRUCTION (code=EXC_I386_INVOP, subcode=0x0)
frame #0: 0x0000000103493b6e Python`_PyUnicodeWriter_PrepareInternal + 488
Python`_PyUnicodeWriter_PrepareInternal:
-> 0x103493b6e <+488>: bextrl %ecx, 0x20(%r14), %ecx
0x103493b74 <+494>: movq 0x10(%r14), %rdx
0x103493b78 <+498>: movl %ecx, 0x10(%rbx)
0x103493b7b <+501>: movq %rdx, 0x18(%rbx)
Target 0: (Scopy) stopped.
(lldb)

This is from the MacOS crash log:

Crashed Thread: 0 Dispatch queue: com.apple.main-thread

Exception Type: EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes: 0x0000000000000001, 0x0000000000000000
Exception Note: EXC_CORPSE_NOTIFY

Termination Signal: Illegal instruction: 4
Termination Reason: Namespace SIGNAL, Code 0x4
Terminating Process: exc handler [7859]

Thank you,

Jeff

AlexandraTrifan, 2019-02-18:
Hi, 

We can not reproduce this issue. Could you provide some more details, maybe those can help us identify the issue:
- What Python versions do you have installed on your machine?
- Can you run "locale" or "defaults read -g AppleLocale" in a terminal? This should provide locale information and we are requesting it because the stack trace indicates that Python crashes after running "PyUnicode_DecodeUTF8Stateful" in the initialization phase. 

We will provide a version of Scopy which allows the user to limit the decoding capabilities of the Logic Analyzer. That version should no longer raise any issues related to Python. This is just a workaround until we manage to reproduce the issue and find the sequence that causes it.

We are sorry for the inconvenience.

-Alexandra

jcristine, 2019-02-18:
Python Versions:

macOS:~$ python
python             python2.7-config   python3.7          pythonw
python-config      python3            python3.7-config   pythonw2.7
python2            python3-config     python3.7m
python2.7          python3.6          python3.7m-config

macOS:~$ python
Python 3.7.2 (default, Jan 13 2019, 12:50:01)
[Clang 10.0.0 (clang-1000.11.45.5)] on darwin

macOS:~$ python2
Python 2.7.10 (default, Aug 17 2018, 19:45:58)
[GCC 4.2.1 Compatible Apple LLVM 10.0.0 (clang-1000.0.42)] on darwin

macOS:~$ python3
Python 3.7.2 (default, Jan 13 2019, 12:50:01)
[Clang 10.0.0 (clang-1000.11.45.5)] on darwin

locale:

macOS:~$ locale
LANG="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_CTYPE="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_ALL=

defaults read -g AppleLocale:
macOS:~$ defaults read -g AppleLocale
en_US

I saw another message thread with a similar topic that recommended renaming the "/Applications/Scopy.app/Contents/MacOS/decoders" folder.  I tried that and it did not work.  Scopy still crashed upon clicking "connect" (both USB and TCP/IP).

Thank you,

Jeff

AlexandraTrifan, 2020-09-17:
A lot of things changed on the connection and calibration pieces and a lot of stability issues were addressed (on MacOS included).
Can this be reproduced using the latest build ( https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/35251245/job/oxx7gqvvslv7gf19/artifacts )?

closed 2020-10-19 by adisuciu