analogdevicesinc/scopy #535: Scopy incompatible with Mac OS 10.14.3

gillScott, 2019-02-18:
New with ADALM 2000.  drivers download and install just fine.  Scopy downloads and disc symbol appears on desktop.  Scopy Window opens with Scopy application .  But when attempt to open, get message window saying "check with developer to make sure Scopy works with this version of MacOS"
Is this a known problem?  Is there a new Scopy release available?
Is Alice compatible with OS 10.14?  If I down load Windows version in to Windows 10 on top of Boot Camp will there be a driver problem.  scottG

AlexandraTrifan, 2019-02-19:
Hi,

We just updated Mac OS to 10.14.3 and can not reproduce this issue. You are able to open Scopy up, see the list of devices but after this you get the version related error? This Scopy release was built on a Mac OS 10.13 so it should be compatible with the latest versions. The version issue can only appear on older OS versions.

Did you install Scopy in the /Applications folder or are you running it from /Volumes/Scopy (the location where the .dmg was mounted?)
Can you provide some screenshots with Scopy and the error message?

Regarding Windows on top of Boot Camp, we haven't tested this scenario, but there should not be any issues.

Instructions on how to use Alice on all the major operating systems can be found here: https://wiki.analog.com/university/tools/m2k/alice/users-guide-m2k .

Thank you

-Alexandra

gillScott, 2019-02-22:
Thanks for the reply.  I did try to open Scopy both by putting it in Applications and from the icon that .dmg placed on my desk top.  I get a folder with some Scopy related, user instructions and other items but no application.  I may try to delete all and renown load.

I did successfully install Scopy under Windows 10 on Boot camp.  I also have the circuit simulator Tina there.  But I would like to have it on native OS 10.14.  I will work on the screen shot.
Thanks, I will get back with you.
Scott Gilles

gillScott, 2019-02-22:
Screen shots follow in separate emails do to side limits .

gillScott, 2019-02-22:
Eecond screen shot.  Error msg
Scott Gilles

gillScott, 2019-02-22:
Third screen shot

gillScott, 2019-02-22:
Fourth screen shot

gillScott, 2019-02-22:
Il deleated all my first download of Scopy and associated icons, and re-downloaded   the .dmg.  I dragged the dmg to the desktop then into applications.  the disk image for the dmg Remains in the application window.  When I open that, I get the Scopy icon.  Opening  that gives same error message that I sent in screen shot.  When i right click and select the Unix option, I see very brief log that" library not loaded".  Is there a separate library that has to be loaded?  On Windows every thing was straight forward.
Scott Gilles

AlexandraTrifan, 2019-02-25:
Hi, 

It looks like a dependency was missing from the dmg. We added a fix for this in a new release. Can you try to use this version: https://github.com/analogdevicesinc/scopy/releases/tag/v1.0.4 ?

Thank you, 

-Alexandra

gillScott, 2019-02-26:
Thanks Ms. Trifan,
Worked fine.  I have not run any tests yet on oscillator/function generator but will.  And get on my first activity with the OpAmp.  I had to look up in one of my Linux books to remind my self about  dependency.  I also found the term “Dependency Hell” in Wikipedia, apropo.   That is the problem with becoming a manager. You have a !00+ 30-somethings to worry such things.  Then I have been retired for 10 years trying to be a hobbyist.  It will be fun doing the activities with Scopy, and with my desktop lab instruments.
Thanks,
Scott Gilles

closed 2019-02-27 by adisuciu