analogdevicesinc/scopy #536: The signal generator does not start both channels at the same time

adisuciu, 2019-02-26:
The signal generator does not start both channels at the same time.

FW - 0.15
[image]

Fw - 0.17
[image]

dbogdan, 2019-02-26:
The issue will be solved by implementing gpio_chip->set_multiple() for https://github.com/analogdevicesinc/linux/blob/m2k/drivers/input/keyboard/adp5589-keys.c.

adisuciu, 2019-04-02:
Should be fixed with https://github.com/analogdevicesinc/scopy/pull/557

amiclaus, 2019-05-08:
On Scopy version: e28686c (v1.0.5), FW: v0.23 the bug is fixed.

closed 2019-05-08 by amiclaus