analogdevicesinc/scopy #546: unable to build scopy

pantechsolutions, 2019-03-11:
hello unable to build scopy on windows
[image]

can anyone help

AlexandraTrifan, 2019-03-12:
Hi, 

Can you check the Qt version with "qmake --version"? 
Did you install Qt using these commands: 
` wget http://repo.msys2.org/mingw/x86_64/mingw-w64-x86_64-qt5-5.9.1-1-any.pkg.tar.xz`
 `pacman -U --noconfirm mingw-w64-x86_64-qt5-5.9.1-1-any.pkg.tar.xz`

Can you try to update your Qt using: 
`pacman -S mingw-w64-x86_64-qt5` ?

Thank you

-Alexandra

pantechsolutions, 2019-03-12:
Hello Alendara

Thank you so much for your reply,i follow the same steps as given by you,but still i am not able to make the build .attach is the screenshot,can you share your comments.

Thank you.
[image]

pantechsolutions, 2019-03-12:
[image]

pls find the above screenshot for the cmake results

AlexandraTrifan, 2019-03-12:
Can you check the Qt version with "qmake --version"?
If the Qt version is 5.12, you should copy the pre-compiled dependencies again, using: ` wget "https://ci.appveyor.com/api/projects/analogdevicesinc/scopy-mingw-build-deps/artifacts/scopy-mingw64-build-deps.tar.xz?branch=master&job=Environment: MINGW_VERSION=mingw64, ARCH=x86_64" -O- | tar -C / --strip-components=1 -xJ`

After that, please remove everything from your build folder, in order to start clean.
Run cmake and ninja again and let us know how it works. 

Thank you.

-Alexandra

pantechsolutions, 2019-03-12:
Hello Alendra

finally i was able to make the build,after i set the environmetal variables . I am getting the following error

[image]

[image]

here is the qmake version
qmake --version

$ qmake --version
QMake version 3.1
Using Qt version 5.12.1 in C:/msys64/mingw64/lib
[image]

[image]

pantechsolutions, 2019-03-13:
Hello alexandra ,finally it worked,as the environmental variables  has to be set in the system variables.

Thank you so much for your support.

Is there any documentation available for creating  the build as a windows set up  file. We are using adalm2000 in our product.

Thank you.

AlexandraTrifan, 2019-03-13:
Hi, 

You can find Windows installers on the Release page: https://github.com/analogdevicesinc/scopy/releases .
This is the latest release: https://github.com/analogdevicesinc/scopy/releases/tag/v1.0.3 .

All the documentation for ADALM2000 and Scopy can be found here: https://wiki.analog.com/university/tools/m2k/scopy 
and 
https://wiki.analog.com/university/tools/m2k .

You can also find information about building and creating an installer in the Appveyor file ( https://github.com/analogdevicesinc/scopy/blob/master/appveyor.yml ). Appveyor is a continuous integration service that we use to build Scopy. It uses MSYS2 and MinGW. It can provide useful information. 

Thank you

-Alexandra

pantechsolutions, 2019-03-13:
Thank you for your reply.

I tried to connect with ADALM2000 ,when i click connect the scopy suddenly closes.i have  record a video and sent to you thru engineer zone.

can you have a look at it and reply ,thank you.

here is the screenshot
[image]

[image]

scopy closes immediately after the screenshot.

pantechsolutions, 2019-03-13:
please find the error in the video

https://youtu.be/vIWw2cVRtog

AlexandraTrifan, 2019-03-13:
Does this happen using the Scopy you built locally, or using the release downloaded from the github release page?

-Alexandra

pantechsolutions, 2019-03-13:
it also happens if i install thru the installer you  provided on the link.

i am testing with three laptops with three different adalm2000, i am getting the same error.

In one laptop i have install only the driver and the installer file,but it is crashing,sometimes it display  calibrating and then closed. i am using windows 10 .teh usb cable which was supplied by analog devices.

I have installed the drivers for adalm2000.

is any other info needed,Thank you

AlexandraTrifan, 2019-03-13:
Can you install libiio, using this installer: https://github.com/analogdevicesinc/libiio/releases/download/v0.15/libiio-0.15.g6ecff5d-Windows-setup.exe ? 
After that, open a command prompt, type **iio_info -a** and provide the output please.

-Alexandra

pantechsolutions, 2019-03-13:
[image]

pls find the screen shot

pantechsolutions, 2019-03-13:
[image]

pls chk the usb version.it differs for our build

AlexandraTrifan, 2019-03-14:
Hi, 

Please connect the M2k to your PC before running iio_info -a. It should provide more information about the detected device. Right now it seems that the board was not connected to the PC.

Thank you. 

-Alexandra

pantechsolutions, 2019-03-14:
the device is connected ,I am using windows10 .
[image]

pantechsolutions, 2019-03-14:
is there any possibility to connect the system via any remote desktop software.

That would be of great help,as we have to deliver our adalm2000 kits to our customers.

pantechsolutions, 2019-03-15:
Hello Alexandra
I s there any update,thank you.

AlexandraTrifan, 2019-03-15:
Hi,

Are you connect M2k through a docking station? We've had problems with those before.

Does M2k appear as a mass storage device? 
[image]

Can you open the Device Manager and check if you have all the following entries for M2k?
[image]

Thank you.

-Alexandra

AlexandraTrifan, 2019-03-15:
Could you also try to connect through IP? (documentation here: https://wiki.analog.com/university/tools/m2k/scopy#connecting_to_a_remote_device )

You can do it from Scopy, using the "+" button. You should also run **iio_info -u "ip:192.168.2.1"** and check the result.

AlexandraTrifan, 2019-03-15:
You could also start Scopy using the gdb debugger. Open a Command Prompt. 
Then go to the Scopy folder (you could use the following command: **cd C:\Program Files\Scopy\** . Then run **gdb Scopy.exe** . 

The program will load, then type **run** and press Enter. (you might have to press enter 2 times, because that might happen in command prompt) . 

Scopy will start, a bit slower than usually. After it starts, try to connect to it. It will crash/hang. 

Go into the command prompt and press CTRL+C to get the gdb prompt (something like **(gdb)**). When that's visible, write **bt**, then **bt full** and post the output please.

pantechsolutions, 2019-03-20:
Hello Alexandara

Thank you for your support,we try to build on another laptop and it worked,but not able to debug on the system which was not working.

do you have any documentation to edit the ui file and add some functions on it.

Is there any matlab integartion to the tool?

can we visualize function generator and oscilloscope at the same time.

Thank you

AlexandraTrifan, 2019-03-21:
Hi, 

All the Scopy related documentation can be found here: https://wiki.analog.com/university/tools/m2k/scopy . For the ui we use Qt, so all the related detailed documentation can be found on their official page https://doc.qt.io/ .

There isn't any matlab integration right now. There will be, but we can't give an ETA right now.

You can visualize any tool in parallel. This is described in the documentation ( https://wiki.analog.com/university/tools/m2k/scopy#user_guides ) under the "Detaching instruments" section.

Thank you.

-Alexandra

pantechsolutions, 2019-03-22:
Hello Alexandra,

How to remove this  button.

[image]

so far i sold out 20 ADALM2000 kits ,customers are asking me is the software not completed yet? on seeing this button.

pls let me how to remove this .If they report any bug ,i will report here.

Build is fine,it is working well in appveyor.

What is the use of power USB connector in ADALM2000 ,as i am always using only the USB connection.

adisuciu, 2019-03-22:
Hi,

There is no software without bugs and lots of software(even more mature than Scopy) have this functionality - attached is a Google Chrome screenshot. Having this button makes it easy for the users to get in touch with the developers directly through the user interface, therefore I wouldn't suggest removing it. If you still want to remove it, I'm sure you can figure out how to do it, but we cannot support that version.

You can power the M2k from the USB power connector and use the M2k's OTG port in a similar way you can use it on the pluto: https://wiki.analog.com/university/tools/pluto/usb_otg_host_function_support

-Adrian

pantechsolutions, 2019-03-22:
Hello Adrian
I could understand ,but i am finding very hard to convince my target customer. I have  a target of selling 100 nos of adalm 2000 within next two months.it would be easier for me if i could have the one. i could rebuild and deliver the updated version.
Kindly help me to remove the display.just let me know the steps i could do that

thank you

rgetz, 2019-03-22:
It’s open source - I would suggest if you need to make a fork and change things yourself. We can’t really help with these sorts of things we don’t agree with. 

- Robin

pantechsolutions, 2019-03-23:
Hello
I have installed code composer studio and scopy stops working after that.i have uninstalled code composer studio and reinstall the scopy from the latest release from wiki.

all input channels are not working,pls find the screencast of the error.

https://youtu.be/XLIoL96Cyq4

when i try to measure a voltage using voltmeter it hangs.
i tried to use oscilloscope it hangs.

rgetz, 2019-03-23:
There shouldn’t be any sort of name space clashes between code composer and scopy. 

Did you reboot your PC?

pantechsolutions, 2019-03-24:
yes,i have uninstalled ccs and also reinstallled scopy.

but it was not working,signal generator is working

pantechsolutions, 2019-03-24:
can anyone help me to sort the issue,thanks

rgetz, 2019-03-24:
Someone will need to try to replicate the issue before we can provide any info. 

I would suggest to uninstall both CCS and Scopy, reboot; and then install Scopy.

pantechsolutions, 2019-03-25:
I Have done but still it is not working, i cannot format the system,is there any option to make it work.

adisuciu, 2019-03-25:
CCS should not alter Scopy installation in any way. After uninstalling Scopy make sure all files in Program Files are deleted. Also delete all files in c:\Users\asuciu\AppData\Roaming\ADI\

I can see in the video that you are running Scopy while low on battery. Please try using a power adapter to charge your laptop.
Please connect the M2k directly to the laptop - no docking station - no USB hubs.
Try adding the 192.168.2.1 device and connect to it (it's the same device) and retry the test.

If all else fails, please try running these steps.

> You could also start Scopy using the gdb debugger. Open a Command Prompt.
> Then go to the Scopy folder (you could use the following command: **cd C:\Program Files\Scopy** . Then run **gdb Scopy.exe** [...]

-Adrian

pantechsolutions, 2019-03-26:
Hello adisuciu

It worked after i delete the files from

c:\Users\jeeva\AppData\Roaming\ADI\

Thank you so much.

closed 2019-03-26 by adisuciu