analogdevicesinc/scopy #553: Scopy fails to start in the presence of ~/.config/ADI on Linux

balbirthomas, 2019-03-20:
The following behavior was observed after updating ADALM2000 firmware to version 0.22 and building a recent version of Scopy from git ("scopy -v" reports sha 65d9ea9) . 

- If the directory ```~/.config/ADI``` exists Scopy fails to start, irrespective of weather an   ADALM2000 is connected or not. 
- Deleting the entire ```~/.config/ADI``` directory allows Scopy to start normally again.
- However every time scopy runs it creates the ```~/.config/ADI``` causing the problem again the very next time Scopy is run

balbirthomas, 2019-03-20:
To be more specific it is the presence of ```~/.config/ADI/Scopy.ini``` that is causing this problem

AlexandraTrifan, 2019-03-21:
Hi,

Please try to build the latest master version. I could not find the commit you mentioned to see what changes were done until that point.

You could try to run "gdb scopy", then type "run" and when it crashes, just type "bt full". This should give us more information about the exact place where Scopy fails to start in certain conditions.

Thank you.

-Alexandra

balbirthomas, 2019-03-21:
Hi,
Here is the GDB backtrace obtained after attaching to a Scopy program that failed to launch the GUI.

```
(gdb) bt full
#0  0x00007f498fa3b8bd in poll () at ../sysdeps/unix/syscall-template.S:84
No locals.
#1  0x00007f49922fa9f6 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#2  0x00007f49922fab0c in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#3  0x00007f499086a1cf in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#4  0x00007f4990813d1a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#5  0x00007f499081c46c in QCoreApplication::exec() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#6  0x0000562e25e874e7 in main ()
No symbol table info available.
```

I have also pushed my branch of Scopy to github. Please see https://github.com/balbirthomas/scopy/tree/debian . Also note all changes I have made for packaging on Debian are isolated in the "debian" directory. In particular the only change to Scopy source code is https://github.com/balbirthomas/scopy/commit/ec0c0cc822438aa13118b4c427f59ca98c9154e5 in order to be able to compile it with the Qt 5.7 that ships with Debian/Stretch. 

Do let me know if you need any further information (such as strace output) or would like me to check anything.

AlexandraTrifan, 2019-03-21:
Hi, 

I created a build using Qt 5.7 and tested using the code from your github repo, but I still can't reproduce the issue. 
Could you attach the Scopy.ini? 

Thank you.

-Alexandra

balbirthomas, 2019-03-21:
Hi

[Scopy.ini](https://github.com/analogdevicesinc/scopy/files/2992275/Scopy.txt)
is attached. There is also a Scopy.ini.bak which is the same. 

Regards

balbirthomas, 2019-03-21:
Hi Alexandra,

Does any of Scopy's dependencies such as ```libiio`` use ```Scopy.ini``` if so I can try to debug those.

regards

AlexandraTrifan, 2019-03-21:
Hi,

Only Scopy uses the Scopy.ini file to load some preferences/settings when Scopy starts. I tested using the .ini file you provided, but eveything looks ok. I am testing on Ubuntu, I was thinking it might be some dependency mismatch on your system. 

I am currently working on creating a local build on Debian, with Qt 5.7,  to check this. I will keep you updated on the progress.

Thank you

-Alexandra

balbirthomas, 2019-07-28:
Hi

I can confirm that this issue is no longer reproducible in the new release of Debian (Buster) and recent build of Scopy and its dependencies. 

Regards

closed 2019-07-28 by balbirthomas