analogdevicesinc/scopy #566: network analyzer

pantechsolutions, 2019-04-04:
for analog low pass rc filter , i am getting correct response in network analyzer,but for digital filter the phase is different ,i use a dsp processor.pls find the screenshot.
[image]

pantechsolutions, 2019-04-04:
pls find the screenshot for low pass filter
[image]

pantechsolutions, 2019-04-04:
[image]

adisuciu, 2019-04-05:
Can you confirm that is not the characteristic of the digital filter with another instrument?
It looks like the network analyser works correctly. Maybe your digital filter has a different characteristic from the analog one.

closed 2019-05-02 by adisuciu