analogdevicesinc/scopy #576: <Instrument>: <Short description of the bug>

cobermay, 2019-04-29:
OS Version: Windows 10 (10.0)
FW Version: v0.22
SW Version: 91c4f0c

Description of the bug: External voltage source power on crashes Scopy
Steps to reproduce:

- External Power-Source set to 3,3V (current limited to about 0,3 A)
- Resistor with 27k be
[Scopy.ini.txt](https://github.com/analogdevicesinc/scopy/files/3128337/Scopy.ini.txt)
tween + and -
- ADALM2000 probes on the resistor
- Scopy in Oscilloscope mode or Voltmeter mode
- Power on the Power-Supply in case Scopy doesn't freeze just switch off and on again (Scopy looks like normal but doesn't react to start/stop any more, or doesn't update the values)

The ini files might be useful to reproduce the error.
The ini file is located at: C:/Users/chris/AppData/Roaming/ADI/Scopy.ini
Please consider attaching it.

mbanciso, 2019-05-02:
You also need to connect a GND pin of ADALM2000 to the GND of the external power supply.
Please whether Scopy crashes with proper grounding.

adisuciu, 2019-05-02:
We couldn't reproduce the issue from a SW point of view (no crash). For the oscilloscope instrument, it looks like you are using normal triggering (this will only trigger on power supply on/off). The voltmeter should work fine though. 

Also make sure you connect GND pin correctly as mbanciso mentioned.

-Adrian

closed 2019-07-09 by adisuciu