analogdevicesinc/scopy #593: Noise does not reflect on oscilloscope reading

tagoylo, 2019-05-15:
Scopy version: 6f7e649
fw version: v0.23

Steps to reproduce:

Set ch1 and ch2 to any waveform except square wave
Change Noise to uniform, gaussian, impulse, etc.
Increase Noise amplitude.
Then monitor output on oscilloscope
The waveform displayed on oscillscope does not add the noise. For sine wave wave there is noise added but does not correspond to the amplitude.

I tried monitoring the output on discovery and the noise is added.
[image]

DanielGuramulta, 2019-05-15:
Looks like a duplicate of this: https://github.com/analogdevicesinc/scopy/issues/338. If you go into "Preferences" and disable the  "adc filters", is the added noise still not showing up?

tagoylo, 2019-05-15:
Disabled adc filters and the noise is now showing up. Closing this item as it is invalid.

closed 2019-05-15 by tagoylo