analogdevicesinc/scopy #594: Noise in stairstep seems odd

tagoylo, 2019-05-15:
Scopy version: 6f7e649

Not sure if this is a bug or it's expected. 

When noise is added to a stairstep the following is the result.
[image]

Steps to reproduce: 

Go to signal generator buffer
Load stairstep waveform in .csv file format. 
[signal_generator_buffer_test.zip](https://github.com/analogdevicesinc/scopy/files/3182208/signal_generator_buffer_test.zip)

Then add noise, uniform 1V.

adisuciu, 2019-05-15:
This is the expected behavior. Noise generator adds amplitude on top of each sample. if your input file has few samples(12), this is the expected result.

tagoylo, 2019-05-15:
Noted. Closing this item as it is invalid.

closed 2019-05-15 by tagoylo