analogdevicesinc/scopy #606: Not using position setting for YX plots?

damercer, 2019-05-23:
I just installed the new version 1.0.5 of Scopy. I've not played around with the X/Y plot feature before so this question might not be new to version 1.0.5 but looking at this screen shot it looks like the position setting is not being used for X and Y grid?

[image]
 
If I set up for a plot just using the upper right quadrant ( +/+ ) for positive only signals ( like for a device I/V curve ) the trace is off screen. Based on the grid labels I should see at least see the 0 to +2.5V parts of the signals.

Also it looks like part of the time grid is covered by the X/Y grid when the controls are shown on the right. There is no overlap of the grids when the controls are closed.

Doug

adisuciu, 2019-05-24:
Fixed with https://github.com/analogdevicesinc/scopy/pull/607

amiclaus, 2019-08-30:
On Scopy build: d198b9a (v1.0.6) the bug is fixed.

closed 2019-08-30 by amiclaus